mincequi/cornrow #3: pairing procedure

robaru, 2019-01-11:
Hi again!
I am having issues pairing my Android phone. The app can control the gstreamer EQ but I cannot connect to the bluetooth in order to stream audio.

Is there any procedure or sequence to follow in order to have everything working (audio stream and EQ)?

Thank you again,
Roberto

mincequi, 2019-01-14:
Hi,
which exact Android device do you use? Which distribution do you use on your RasPi?

Sometimes android devices seem to show two devices (the streaming sink AND the eq control sink). You have to connect the correct one. Usually (and working correct on iOS) the eq control device should not show up in the list of pairable devices.

You can also try to restart cornrowd since there might be a system-wide timeout for device pairing.
Also consider restarting the bluez daemon.

With the thirdparty lib bluez-qt there is a test app provided called: _mediaendpointconnector_.
Can you try running this one (cornrowd has to be stopped before) and paste the output here?

robaru, 2019-01-16:
Hi,
I am using an Android Xperia L1 with Android 7. The distribution is Raspian (updated).

Your daemon is showing the streaming sink occasionally while more often I can detect the eq control.

By checking the bluetooth status I obtain:

```
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2019-01-11 12:18:47 GMT; 5 days ago
     Docs: man:bluetoothd(8)
 Main PID: 357 (bluetoothd)
   Status: "Running"
   CGroup: /system.slice/bluetooth.service
           └─357 /usr/lib/bluetooth/bluetoothd

Jan 11 12:18:47 raspberrypi bluetoothd[357]: Bluetooth management interface 1.14 initialized
Jan 11 12:18:47 raspberrypi bluetoothd[357]: Failed to obtain handles for "Service Changed" characteristic
Jan 11 12:18:47 raspberrypi bluetoothd[357]: Sap driver initialization failed.
Jan 11 12:18:47 raspberrypi bluetoothd[357]: sap-server: Operation not permitted (1)
Jan 11 12:18:48 raspberrypi bluetoothd[357]: Failed to set privacy: Rejected (0x0b)
Jan 11 12:18:48 raspberrypi bluetoothd[357]: Endpoint registered: sender=:1.3 path=/MediaEndpoint/Sink/Sbc
Jan 16 21:53:01 raspberrypi bluetoothd[357]: /org/bluez/hci0/dev_D4_38_9C_31_7F_65/fd0: fd(22) ready
Jan 16 21:55:57 raspberrypi bluetoothd[357]: Endpoint unregistered: sender=:1.3 path=/MediaEndpoint/Sink/Sbc
Jan 16 21:56:04 raspberrypi bluetoothd[357]: Endpoint registered: sender=:1.12 path=/MediaEndpoint/Sink/Sbc
Jan 16 22:04:00 raspberrypi bluetoothd[357]: Endpoint unregistered: sender=:1.12 path=/MediaEndpoint/Sink/Sbc
```
 
About the bluez test: can you point me out how to find the compiled file? 

Thank you so much for the support. 

Roberto

mincequi, 2019-01-18:
Hi,
did you also update the bluez daemon? AFAIR, the bluez daemon was heavily patched for RasPi and a newer version (from debian testing) might not work.

What i can see from the logs is, that the Endpoint gets unregistered (in the last line).
You can also try to start my daemon as regular app (not in daemon mode). This might need super user privileges.

Within cornrowd, i disabled building tests for the bluez-qt lib. You need to get it from https://github.com/KDE/bluez-qt and build the whole library and within your build folder you will find the test binary.

Btw. did you manage to get another bluetooth audio sink up and running?

Sorry for inconvenience, i guess i need to add some more diagnostics, like logging, to my daemon.

closed 2019-10-14 by mincequi