mincequi/cornrow #4: iPhone unable to control volume

jesper216, 2019-01-29:
Hi,

Successfully installed on RaspberryPi B3+. When my macbook is connected, I can change the volume with my macbook. However with my iPhone connected, changing the volume on the iPhone doesn't change the volume on the speaker.

Another thing, the speaker isn't discoverable automatically. Would it be possible to make it discoverable automatically when no device is connected?

Thanks in advance & best regards,
Jesper

jesper216, 2019-01-30:
Thanks for your reply. I'll eagerly await the volume control function!

I was able to connect by manually setting discoverable mode (bluetoothctl --> discoverable on). Since talked about a discoverable timeout, should it have been discoverable at some point without me manually making it discoverable?

Thing is, I'd like to be able to easily connect a new device, so disconnecting one device and making the daemon discoverable for other (new) devices. Maybe adding buttons to my Pi would be the way to go (e.g. "disconnect current" and "turn on discoverability").

mincequi, 2019-01-30:
My daemon makes the bluetooth sink discoverable immediately after start-up. Depending on your system's setting (/etc/bluetooth/main.conf) this might timeout (default after 3 minutes).
I will try to overwrite this default value with a patch, but it might not be allowed to overwrite system's defaults.

jesper216, 2019-01-30:
Oh I missed that, then maybe I just waited too long. I'll check it out when
I'm home in a couple of hours. If the device is discoverable for a couple
of minutes after reboot that is fine. Thanks for the information.

mincequi, 2019-10-14:
Volume Control is implemented on latest master

closed 2019-10-14 by mincequi