mincequi/cornrow #5: Building of Pipeline.cpp fails

Mladia, 2019-07-01:
I get an error, when trying to build the project. This what I have cloned `git clone --branch v0.1.0 --recursive https://github.com/mincequi/cornrow`

```
[ 22%] Building CXX object cornrowd/src/audio/CMakeFiles/audio.dir/Pipeline.cpp.o
/home/niki/Repos/cornrow/cornrowd/src/audio/Pipeline.cpp: In member function ‘void audio::Pipeline::setCrossover(const common::Filter&)’:
/home/niki/Repos/cornrow/cornrowd/src/audio/Pipeline.cpp:114:18: error: ‘class GstDsp::Crossover’ has no member named ‘setFrequency’; did you mean ‘frequency’?
     m_crossover->setFrequency(crossover.f);
                  ^~~~~~~~~~~~
cornrowd/src/audio/CMakeFiles/audio.dir/build.make:110: recipe for target 'cornrowd/src/audio/CMakeFiles/audio.dir/Pipeline.cpp.o' failed
make[2]: *** [cornrowd/src/audio/CMakeFiles/audio.dir/Pipeline.cpp.o] Error 1
CMakeFiles/Makefile2:537: recipe for target 'cornrowd/src/audio/CMakeFiles/audio.dir/all' failed
make[1]: *** [cornrowd/src/audio/CMakeFiles/audio.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

mariusdkm, 2019-09-10:
Hey, I get the same error but while i run make.
Did you get an workaround yet?

mincequi, 2019-10-14:
Please checkout recent master. It should work again.

closed 2019-10-16 by mincequi