mincequi/cornrow #6: Control via python script

TGWC, 2019-07-30:
Hi,

It's possible to control via a python script? This is,  to start/stop, and to just be discoverable when I click a button? This is to avoid random users to connect to BT, they should have to press a  button first to pair...

Thanks a lot

mincequi, 2019-10-08:
Hi,
currently, there are no plans for such a feature. However, i can check whether I could implement any kind of socket interface.

mincequi, 2019-10-18:
Hi,
btw, what kind of button? Do you mean a physical one? Supporting GPIO should be easy.

abelmatser, 2021-12-28:
Or perhaps include a method to allow users to connect within a set timeframe, let's say 1 minute.

mincequi, 2021-12-28:
hmm, still not sure, how we want to interface with cornrow.
Maybe, it is enough to add a PIN code and make it configurable via config file. What do you think?

mincequi, 2022-01-02:
@abelmatser 
Yes, this sounds acceptable to me. If you could even help out with a PR this would be best. I am not quite sure, when i find the time to do this. :)

Also great to hear, that someone is actually using the App. So, waht's your experience with it?

closed 2022-11-07 by mincequi