mincequi/cornrow #7: Cannot connect with MacBook Pro

arbego, 2019-12-01:
Hello, when I try to connect from my MacBook Pro the daemon crashes with following message:

[image]

Apologies for the crappy photo, the Pi was connected to my TV.

Good work by the way, it pretty well worked right after I installed the dependencies and the .deb package.

mincequi, 2019-12-01:
Hi, thanks for your feedback. I can reproduce this with my MacBook Pro and raspberry pi.
This occurs at AudioConverter.cpp::37 from += size(conf.codec);
Seems to be an issue when doing pointer arithmetic.

Under investigation.

mincequi, 2019-12-01:
This is fixed with v0.5.1.
Please verify.
Thanks :)

closed 2019-12-12 by mincequi