mincequi/cornrow #10: Build problem

macek-jm, 2020-03-02:
Can you help me.
after 82% dpkg-buildpackage returns error
the last of make:

[ 82%] Built target KF5BluezQt
make[3]: Leaving directory '/home/pi/cornrow/obj-arm-linux-gnueabihf'
make[2]: *** [Makefile:133: all] Error 2
make[2]: Leaving directory '/home/pi/cornrow/obj-arm-linux-gnueabihf'
dh_auto_build: cd obj-arm-linux-gnueabihf && make -j4 returned exit code 2
make[1]: *** [debian/rules:9: override_dh_auto_build] Error 2
make[1]: Leaving directory '/home/pi/cornrow'
make: *** [debian/rules:5: build] Error 2
dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2

mincequi, 2020-03-02:
Hi, thanks for your feedback.
What‘s your build environment?
Are there more logs you can provide?

macek-jm, 2020-03-02:
Thanks for reply, 
its Rpi
Linux musicbox 4.19.50-v7+ #896 SMP Thu Jun 20 16:11:44 BST 2019 armv7l GNU/Linux
Copyright (c) 2012 Broadcom
version a59fb7a74180be0111dbc5c18a37ec6df86f14a3 (clean) (release) (start_cd)
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster

where to find logs,
dpkg.log ?

mincequi, 2020-03-03:
Apparently, there had been issues with the Debian scripts. I did a quick fix. Please check again :)
Thanks a lot.

macek-jm, 2020-03-07:
today i tried one more time, but is the same

[ 82%] Built target KF5BluezQt
make[3]: Leaving directory '/home/pi/cornrow/obj-arm-linux-gnueabihf'
make[2]: *** [Makefile:133: all] Error 2
make[2]: Leaving directory '/home/pi/cornrow/obj-arm-linux-gnueabihf'
dh_auto_build: cd obj-arm-linux-gnueabihf && make -j4 returned exit code 2
make[1]: *** [debian/rules:9: override_dh_auto_build] Error 2
make[1]: Leaving directory '/home/pi/cornrow'
make: *** [debian/rules:5: build] Error 2
dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2

mincequi, 2020-03-07:
Could you follow the non-debian Installation steps?
Using cmake (not dpkg-buildpackage)?

macek-jm, 2020-03-08:
ok now we can see much more

[ 75%] Building CXX object cornrowd/CMakeFiles/cornrowd.dir/cornrowd_autogen/mocs_compilation.cpp.o
[ 76%] Linking CXX executable cornrowd
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libavcodec.so: undefined reference to `bcm_host_is_fkms_active'
collect2: error: ld returned 1 exit status
make[2]: *** [cornrowd/CMakeFiles/cornrowd.dir/build.make:129: cornrowd/cornrowd] Error 1
make[1]: *** [CMakeFiles/Makefile2:197: cornrowd/CMakeFiles/cornrowd.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

mincequi, 2020-03-09:
Thanks again for testing.

Your are using an old kernel version(4.19.50-v7), that does not match you libavcodec and/or raps firmware installation. So, you will also get this error when using libavcodec in another application.
Consider upgrading your kernel to at least 4.19.73. 

See also: https://github.com/raspberrypi/firmware/issues/1319

Btw:
Why do you compile yourself? There are binary packages available for pi (arm).

macek-jm, 2020-03-14:
Yeah, now is ok. Was my oversight.

i compile by myself for self-progress ;)

thnx a lot.

closed 2020-03-14 by macek-jm