mincequi/cornrow #12: tested on Pi 4

UrielCh, 2020-04-15:
Hi,

I setup cornrow on a Pi4 from the deb file.
- The I stream 48KHz audio to It with a scream service on a windows machine and it works. YESS.

But:
- I tried the android corocomo, but it do not detect anything (using Bluetooth I think)
- The Android application do not allow me to give the IP of my cornrowd server.
- The listen port can not be change (UDP 4010) and for me had been find using a netstat.
- I would like stream some linux sound to It, but I do not know what protocols is supported, what king of UDP data I should send.

Please add some doc / link to compatible software.
What is this software should be able of.
I do not have any STD error about Bluetooth at startup.

I started to looks for answer In sources, and I build time it's looks like libqt5websockets5-dev should in install on debian base OS

`sudo apt-get install libqt5websockets5-dev`

Can you add a not like 'do not forget to pull the git submodule' ?

mincequi, 2020-04-15:
Hi,

thanks a lot for your feedback.

Coming to your points:
- Yes, BluetoothLowEnergy is shitty on Android. That's why i started a Tcp/Ip based communication (which is not finished yet).
- The Android Application found on PlayStore does not support Tcp/Ip currently. I hope to give an updated very soon. Do you use the version on PlayStore or compile from source?
- Scream always uses port 4010. So, i didn't feel the need to change that. However, if this is required, i can realise that.
- cornrowd currently acts as Bluetooth and Scream audio sink. No further protocols are supported as of now. If you can manage to connect your linux PC via Bluetooth to cornrowd, this would solve it. Actually, my plan is to also add Airplay support.

Yes, i can add some doc. However, this is still in development and doc might lag behind implementation. This is also true for the websocket requirement. This is also why i recommend to use official release tarballs instead of recent master (as master might break from time to time).

I documented the --recursive option for git pull. This will also pull submodules.

In Short:
- Cornrowd is (primarily) a bluetooth receiver. I added scream as additional audio source. Airplay will come as well.
- Cornrowd has a built-it equalizer which should be controlled via BluetoothLowEnergy. However this does not work in a stable way for Android, so i started implementing a Tcp/Ip backend (using websockets).

What do you want to achieve?
Stream audio from Windows and Linux to your raspberry pi?
Do you want that equalizer functionality?

UrielCh, 2020-04-15:
For my usage:

I have 2-3 device per room, and only one Speaker. So I looking for a solution to avoid having one speaker peer device.

I do not wan any non essential Wireless connection, So I will not use Bluetooth to connect my PC to my Raspberry Pis.

For the git sub-modules Github do not highlight the *.gitmodules* and project using sub modules are very rare. 

may be you can add something like
### Build from Git source (no commanded)

```
apt-get install -y ......
git clone --recursive ....
....
```bash

------

For this Covi-19 season, I'm trying to build the ultimate PC (software speaking).
For the moment it's based on some RYZEN 3 PC with a single graphique card running Proxmox OS.
In each PC I install 1 or 2 linux, 1 windows, 1 Mac OS X, a samba server, a nfs server, a docker...
then with a hotkey keybord, I press a button for each OS. (the VM will get the GFX as passthrough)

The passthrough of the GFX sound card not working fine.
On some of my Proxmox I can have up to 60 Windows VM that will be fun to get all the 60 windows Sound on the same speaker :)

- I do not need any equalizer functionality, having sound is already a nice feature.

I can not help you in C++, I stop C stuff long time ago. I do not like C style linkage and macro.
I have just discover than C++11 / C++14 / C++17 was some king of standard of C lib.

Maybe you should add something like that in your readme.md

## cornrow server
| Plateforme | feature |  Status | 
|---|-----|-----|
| raspberry 3B / 3B+ | Bluetooth GW | :heavy_check_mark: |
| raspberry 4 | Bluetooth GW | :x: BluetoothLowEnergy is shitty |
| raspberry  | Scream client | :heavy_check_mark: |

## corocomo GUI
| Plateforme | feature |  Status | 
|---|-----|-----|
| Android  | connect BT | :heavy_check_mark: |
| Android  | connect TCP | :wavy_dash: in Progress |

mincequi, 2020-04-20:
Maybe pulseaudio is a solution for your purpose. It is the only (networked) sound daemen that runs on all platforms.

How would you handle 3rdParty dependencies (when not using git submodules)?

UrielCh, 2020-04-27:
after some more test,

It's not possible to cast sound from multiple source to the same cornrow.

mincequi, 2020-04-27:
Well, all scream sources are broadcasting to the same multicast address. So, this is (primarily) not an issue with cornrow, but scream itself. From what i understand, scream aims to be a single source / multiple sinks design. You could ask that question to the scream developer.

I personally plan to support/implement a (unicast) RTP sink using AC3 encoded frames. This could (theoretically) support multiple sources with a single sink.

UrielCh, 2020-04-27:
I use my scream in unicast mode, on day I will try multicast.
can you tell me in which file I may apply change to fix that ? in my case all the PC steaming in 48Hz, maybe a can just sum some values.

mincequi, 2020-04-27:
Really? But cornrow only listens to multicast addresses...

You can have a look at ScreamSource.cpp in the libcornrow repo.
We would need to distinguish the packets from each sender and mix them together in a single audio buffer. Might not be so easy, but definitely feasible.

closed 2020-05-29 by mincequi