mincequi/cornrow #13: rpi zero conection issues

jwittke, 2020-04-24:
I installed the newest version of cornrow on my Raspberry PI Zero running DietPI.
Everything works smoothly and I can see the Pi as bluetooth device. When I want to connect to
the raspi, the connection suddenly drops.
```
root@DietPi:~# bluetoothctl
Agent registered
[CHG] Device B4:9D:0B:95:89:E3 Connected: yes
[CHG] Controller B8:27:EB:80:70:84 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:80:70:84 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:80:70:84 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:80:70:84 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:80:70:84 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:80:70:84 SupportedInstances: 0x05
[CHG] Controller B8:27:EB:80:70:84 ActiveInstances: 0x00
[CHG] Controller B8:27:EB:80:70:84 Class: 0x00000000
[CHG] Controller B8:27:EB:80:70:84 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:80:70:84 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:80:70:84 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:80:70:84 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:80:70:84 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:80:70:84 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:80:70:84 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:80:70:84 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:80:70:84 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:80:70:84 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Device B4:9D:0B:95:89:E3 Connected: no
[CHG] Device B4:9D:0B:95:89:E3 Connected: yes
[CHG] Device B4:9D:0B:95:89:E3 Connected: no
[CHG] Device B4:9D:0B:95:89:E3 Connected: yes
[CHG] Device B4:9D:0B:95:89:E3 Connected: no

```
When monitoring on the pi with bluetoothctl I can see the device which wants to connect, but two second later the connection breaks down and bluetoothctl reports it. 

Any hints how to resolve this?

mincequi, 2020-04-25:
Which version did you try? Or Compiled by yourself?
Just tried https://github.com/mincequi/cornrow/releases/download/v0.7.0/cornrowd-rpi_0.7.0_armhf.deb on my PiZero with stock raspbian. Should work though.

What happens if you stop the daemon and manually start it (to see logs):
```
sudo systemctl stop cornrowd.service
sudo cornrowd
```

jwittke, 2020-04-25:
Yes, I used the cornrowd-rpi_0.7.0_armhf.deb you provided. 
Here is an output when starting it directly.
```
root@DietPi:~# sudo cornrowd
time         [ thread name/id ]                   file:line     v|                                                                                                                           
                                                                                                                                                                                             
                                                                                                                                                                                             
12:53:52.593 [main thread     ]     AudioSinkAgent.cpp:64    INFO| Service 0000110D-0000-1000-8000-00805F9B34FB accepted from BQ Aquaris X Pro                                               
Transport state: BluezQt::MediaTransport::State(Pending)                                                                                                                                     
Transport state: BluezQt::MediaTransport::State(Active)                                                                                                                                      
operator() fd:  13 mtu read: 672 mtu write: 1008                                                                                                                                             
12:54:01.747 [main thread     ]         Controller.cpp:158   INFO| set transport> fd: 14, blocksize: 672, rate: 44100                                                                        
12:54:01.748 [main thread     ]       CoroPipeline.cpp:85    INFO| Stop Bluetooth source                                                                                                     
12:54:01.750 [main thread     ]       CoroPipeline.cpp:152   INFO| AppSource stopped                                                                                                         
12:54:01.751 [main thread     ]FileDescriptorSource.cp:25    INFO| opening file descriptor: 14                                                                                               
12:54:01.752 [main thread     ]       CoroPipeline.cpp:77    INFO| Start Bluetooth source
12:54:01.753 [main thread     ]       CoroPipeline.cpp:166   INFO| Starting AppSource
cornrowd: /home/pi/cornrow/thirdparty/libcornrow/src/pi/audio.c:120: audioplay_create: Assertion `error == OMX_ErrorNone' failed.

Loguru caught a signal: SIGABRT                                                                                                                                                              
Stack trace:                                                                                                                                                                                 
0   0xb4dfa130 __default_rt_sa_restorer + 0                                                                                                                                                  
12:54:02.276 [main thread     ]                       :0     FATL| Signal: SIGABRT
Aborted

mincequi, 2020-04-25:
Thanks for your feedback. That's already helpful.
Well, DietPi seems to use different libraries/versions than raspbian does. The issue is related to HDMI audio output on PI. Maybe, the HDMI device is blocked from another process?

You can try this version (using regular alsa output) for now:
https://github.com/mincequi/cornrow/releases/download/v0.7.0/cornrowd_0.7.0_armhf.deb

Once i find the time to try DietPi, i will do that and try to reproduce your issue.

jwittke, 2020-04-25:
Using the regular version cornrowd_0.7.0_armhf.deb now works on my pi zero. 
The only thing I have to mention is, I needed to install the avahi daemon with

`sudo apt-get install avahi-daemon `

too. 
Tank you!

closed 2020-04-25 by jwittke