mincequi/cornrow #14: signal=SEGV

sphh, 2020-09-19:
I just installed `cornrowd_0.7.0_armhf.deb` (BTW there is an easier way to install the deb file with all dependencies: `sudo apt install ./cornrowd_0.7.0_armhf.deb`) on my Raspberry Pi 1B. I did not `unmask` the `cornrowd.service`, but just started it with
```sh
$ sudo systemctl start cornrowd.service
```
to test it. Unfortunately I get the following error:
```
$ systemctl status cornrowd.service
● cornrowd.service - Cornrow daemon
   Loaded: loaded (/lib/systemd/system/cornrowd.service; disabled; vendor preset: enabled)
   Active: failed (Result: signal) since Sat 2020-09-19 20:47:05 CEST; 18s ago
  Process: 721 ExecStart=/usr/bin/cornrowd (code=killed, signal=SEGV)
 Main PID: 721 (code=killed, signal=SEGV)

Sep 19 20:47:02 mopidy systemd[1]: Started Cornrow daemon.
Sep 19 20:47:05 mopidy cornrowd[721]: time         [ thread name/id ]                   file:line     v|
Sep 19 20:47:05 mopidy cornrowd[721]: Loguru caught a signal: SIGSEGV
Sep 19 20:47:05 mopidy cornrowd[721]: Stack trace:
Sep 19 20:47:05 mopidy cornrowd[721]: 0   0xb4e83130 __default_rt_sa_restorer + 0
Sep 19 20:47:05 mopidy cornrowd[721]: 20:47:05.719 [main thread     ]                       :0     FATL| Signal: SIGSEGV
Sep 19 20:47:05 mopidy systemd[1]: cornrowd.service: Main process exited, code=killed, status=11/SEGV
Sep 19 20:47:05 mopidy systemd[1]: cornrowd.service: Failed with result 'signal'.
```

Is this, because I use a Raspberry Pi **1** and the `.deb` file was compiled for Raspberry Pi **2+**?

sphh, 2020-09-20:
I now compiled `cornrow` on my Raspberry Pi 1 (it took about 8 hours!), uninstalled the old one and installed the freshly compiled one. Same result.

If I start `cornrow` from the command line, I get
```
$ /usr/bin/cornrowd
time         [ thread name/id ]                   file:line     v| 

Loguru caught a signal: SIGSEGV
Stack trace:
0   0xb4f01130 __default_rt_sa_restorer + 0
10:10:40.457 [main thread     ]                       :0     FATL| Signal: SIGSEGV
Segmentation fault
```

mincequi, 2020-09-22:
Hi,
Thanks for checking this out. Can you try to run cornrowd from command line without init script?
`cornrowd`

It should print some log messages...

mincequi, 2020-09-22:
Ok, sorry. I just saw you were trying this.

Well, in this case you can try to start it using gdb.
There might be a valid stack trace.

sphh, 2020-09-22:
Here we go:
```sh
$ sudo apt install ./cornrowd_0.7.0_armhf.deb ./cornrowd-dbgsym_0.7.0_armhf.deb
$ gdb -q /usr/bin/cornrowd
Reading symbols from /usr/bin/cornrowd...Reading symbols from /usr/lib/debug/.build-id/9e/9a06b41fade2059cc4d496eab3771a3a22740b.debug...done.
done.
(gdb) run
Starting program: /usr/bin/cornrowd 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
time         [ thread name/id ]                   file:line     v| 
[New Thread 0xb05b30f0 (LWP 30877)]

Thread 1 "cornrowd" received signal SIGSEGV, Segmentation fault.
BluezQt::Adapter::media (this=0x0) at ./thirdparty/bluez-qt/src/adapter.cpp:161
161	./thirdparty/bluez-qt/src/adapter.cpp: No such file or directory.
(gdb) backtrace
#0  BluezQt::Adapter::media (this=0x0)
    at ./thirdparty/bluez-qt/src/adapter.cpp:161
#1  0x00025c80 in bluetooth::Controller::Controller (this=0x1dae728, 
    parent=<optimized out>)
    at /usr/include/arm-linux-gnueabihf/qt5/QtCore/qsharedpointer_impl.h:312
#2  0x0001eee0 in Controller::Controller (this=0x1db5ed0, 
    parent=<optimized out>) at ./cornrowd/src/Controller.cpp:11
#3  0x0001ceac in main (argc=<optimized out>, argv=<optimized out>)
    at ./cornrowd/src/main.cpp:81
(gdb) 
```
Does that mean anything to you?

mincequi, 2020-09-23:
Do you have a recent version of Bluez installed?
Can you check if you find the media interface for your bluetooth adapter?

'sudo dbus-send --system --print-reply --type=method_call --dest='org.bluez' '/' org.freedesktop.DBus.ObjectManager.GetManagedObjects'

Search for 'org.bluez.Media1'

sphh, 2020-09-23:
The `bluez` package installed has the version `5.50-1.2~deb10u1+rpt2`.

```
$ sudo dbus-send --system --print-reply --type=method_call --dest='org.bluez' '/' org.freedesktop.DBus.ObjectManager.GetManagedObjects
method return time=1600879826.907833 sender=:1.6 -> destination=:1.13 serial=16 reply_serial=2
   array [
      dict entry(
         object path "/org/bluez"
         array [
            dict entry(
               string "org.freedesktop.DBus.Introspectable"
               array [
               ]
            )
            dict entry(
               string "org.bluez.AgentManager1"
               array [
               ]
            )
            dict entry(
               string "org.bluez.ProfileManager1"
               array [
               ]
            )
            dict entry(
               string "org.bluez.HealthManager1"
               array [
               ]
            )
         ]
      )
   ]
```
so no `Media` anywhere in sight.

Oh, btw, do I have to pair a device before I call `cornrowd`?

mincequi, 2020-09-24:
Thats looks strange. Is bluetooth actually enabled on your raspi?
I cannot see any bluetooth adapter.

It should look something like this:
`dict entry(
         object path "/org/bluez"
         array [
            dict entry(
               string "org.freedesktop.DBus.Introspectable"
               array [
               ]
            )
            dict entry(
               string "org.bluez.AgentManager1"
               array [
               ]
            )
            dict entry(
               string "org.bluez.ProfileManager1"
               array [
               ]
            )
            dict entry(
               string "org.bluez.HealthManager1"
               array [
               ]
            )
         ]
      )
      dict entry(
         object path "/org/bluez/hci0"
         array [
            dict entry(
               string "org.freedesktop.DBus.Introspectable"
               array [
               ]
            )
            dict entry(
               string "org.bluez.Adapter1"
               array [
                  dict entry(
                     string "Address"
                     variant                         string "B8:27:EB:4B:58:63"
                  )
`

There is no need to pair a device before starting cornrowd.

sphh, 2020-09-24:
Looks like my Bluetooth dongle is not recognized (https://www.raspberrypi.org/forums/viewtopic.php?t=138145):
```sh
$ systemctl status bluetooth
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset
   Active: active (running) since Thu 2020-09-24 21:15:41 CEST; 50s ago
     Docs: man:bluetoothd(8)
 Main PID: 8344 (bluetoothd)
   Status: "Running"
   CGroup: /system.slice/bluetooth.service
           └─8344 /usr/lib/bluetooth/bluetoothd
```
So that looks ok. Also
```
$ dmesg | grep Blue
[   21.102716] Bluetooth: Core ver 2.22
[   21.102928] Bluetooth: HCI device and connection manager initialized
[   21.102975] Bluetooth: HCI socket layer initialized
[   21.103005] Bluetooth: L2CAP socket layer initialized
[   21.103071] Bluetooth: SCO socket layer initialized
[   42.519478] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   42.519505] Bluetooth: BNEP filters: protocol multicast
[   42.519551] Bluetooth: BNEP socket layer initialized
```
But
```sh
$ hcitool dev
Devices:
```
draws a blank. Also
```sh
$ sudo bluetoothctl
Agent registered
[bluetooth]# devices
No default controller available
```
I guess my dongle is too modern for the Raspbian Buster. :-( I got it working without any fuss on Linux Mint 19.3, so I assumed that it will also work on the RPi …

mincequi, 2020-09-26:
Phew, no idea.

Anyhow, i could add some logging to cornrowd, whether an adapter is present...

sphh, 2020-09-26:
Yes, such a log message would be a good idea.

closed 2021-06-22 by mincequi