mincequi/cornrow #16: crashing with "Assertion `error == OMX_ErrorNone' failed."

Rulasmur, 2021-05-03:
`May 03 18:08:07 xxx cornrowd[17384]: 18:08:07.134 [main thread     ]         SbcDecoder.cpp:73    WARN| Sync byte incorrect
May 03 18:08:07 xxx cornrowd[17384]: 18:08:07.146 [main thread     ]         SbcDecoder.cpp:42    WARN| RTP header invalid
May 03 18:08:08 xxx cornrowd[17384]: 18:08:08.231 [main thread     ]FileDescriptorSource.cp:54    WARN| cannot estimate number of slices
May 03 18:08:08 xxx cornrowd[17384]: 18:08:08.234 [main thread     ]         SbcDecoder.cpp:73    WARN| Sync byte incorrect
May 03 18:08:09 xxx cornrowd[17384]: 18:08:09.165 [main thread     ]FileDescriptorSource.cp:54    WARN| cannot estimate number of slices
May 03 18:08:09 xxx cornrowd[17384]: 18:08:09.165 [main thread     ]         SbcDecoder.cpp:73    WARN| Sync byte incorrect
May 03 18:08:09 xxx cornrowd[17384]: 18:08:09.187 [main thread     ]FileDescriptorSource.cp:54    WARN| cannot estimate number of slices
May 03 18:08:09 xxx cornrowd[17384]: 18:08:09.187 [main thread     ]         SbcDecoder.cpp:42    WARN| RTP header invalid
May 03 18:08:10 xxx cornrowd[17384]: 18:08:10.113 [main thread     ]FileDescriptorSource.cp:54    WARN| cannot estimate number of slices
May 03 18:08:10 xxx cornrowd[17384]: 18:08:10.114 [main thread     ]         SbcDecoder.cpp:73    WARN| Sync byte incorrect
May 03 18:08:10 xxx cornrowd[17384]: 18:08:10.125 [main thread     ]FileDescriptorSource.cp:54    WARN| cannot estimate number of slices
May 03 18:08:10 xxx cornrowd[17384]: 18:08:10.125 [main thread     ]         SbcDecoder.cpp:42    WARN| RTP header invalid
May 03 18:08:11 xxx cornrowd[17384]: 18:08:11.846 [main thread     ]FileDescriptorSource.cp:54    WARN| cannot estimate number of slices
May 03 18:08:11 xxx cornrowd[17384]: 18:08:11.847 [main thread     ]         SbcDecoder.cpp:73    WARN| Sync byte incorrect
May 03 18:08:11 xxx cornrowd[17384]: 18:08:11.867 [main thread     ]FileDescriptorSource.cp:54    WARN| cannot estimate number of slices
May 03 18:08:11 xxx cornrowd[17384]: 18:08:11.867 [main thread     ]         SbcDecoder.cpp:42    WARN| RTP header invalid
May 03 18:08:12 xxx cornrowd[17384]: 18:08:12.476 [main thread     ]FileDescriptorSource.cp:54    WARN| cannot estimate number of slices
May 03 18:08:12 xxx cornrowd[17384]: 18:08:12.477 [main thread     ]         SbcDecoder.cpp:73    WARN| Sync byte incorrect
May 03 18:08:18 xxx cornrowd[17384]: 18:08:18.195 [main thread     ]FileDescriptorSource.cp:54    WARN| cannot estimate number of slices
May 03 18:08:18 xxx cornrowd[17384]: 18:08:18.196 [main thread     ]         SbcDecoder.cpp:73    WARN| Sync byte incorrect
May 03 18:08:18 xxx cornrowd[17384]: 18:08:18.227 [main thread     ]FileDescriptorSource.cp:54    WARN| cannot estimate number of slices
May 03 18:08:18 xxx cornrowd[17384]: 18:08:18.227 [main thread     ]         SbcDecoder.cpp:56    WARN| Fragmented packet(s) not supported
May 03 18:08:19 xxx cornrowd[17384]: cornrowd: /home/pi/cornrow/thirdparty/libcornrow/src/pi/audio.c:298: audioplay_play_buffer: Assertion `error == OMX_ErrorNone' failed.
May 03 18:08:19 xxx cornrowd[17384]: Loguru caught a signal: SIGABRT
May 03 18:08:19 xxx cornrowd[17384]: Stack trace:
May 03 18:08:19 xxx cornrowd[17384]: 0   0x74e96130 __default_rt_sa_restorer + 0
May 03 18:08:19 xxx cornrowd[17384]: 18:08:19.282 [main thread     ]                       :0     FATL| Signal: SIGABRT
`

mincequi, 2021-05-03:
Hmm, there seems something wrong with your stream in general.
you could try the non-pi version. There also seems something wrong with your raspbian installation.

Which exact version do you use?

Rulasmur, 2021-05-04:
I admit I had to do a lot of fiddling to get it working at all - and I had bt-speaker installed before this as well. The stream plays for 5-10 minutes before cutting out normally, and then cornrowd crashed with the above - starting the service again makes it work again.

I did also install bluealsa, although I cannot remember if it was for this or for bt-speaker. I'll give the non-rpi version a go

version: cornrowd-rpi_0.7.0_armhf.deb

mincequi, 2021-05-04:
Ok, your experience would be highly appreciated, so that I can improve this tool. The raspi version uses the OMX audio output plugin, which is not so stable at the moment. So, falling back to the plain version of cornrowd might give you a better experience.

Excited for your feedback.

Rulasmur, 2021-05-05:
cornrowd_0.7.0_armhf.deb
worked perfectly, the service hasn't failed on me yet after testing for a while.

I'll close the issue for now

closed 2021-05-05 by Rulasmur