mincequi/cornrow #17: Snapcast integration - output to ALSA loopback, FIFO, or STDOUT?

eriqjaffe, 2021-09-03:
I'd love to be able to integrate this with Snapcast as it's the only Bluetooth audio daemon I've found that works reliably on a Raspberry Pi.  Snapcast can read from multiple source types but I'm unclear how to redirect cornrow's output to any of them.  Any pointers would be appreciated!

mincequi, 2021-09-04:
Is Snapcast a bluetooth receiver on its own? Never heard of it...

Well, one could implement such an integration. Shouldn't be too hard.

eriqjaffe, 2021-09-05:
Snapcast is a multi-room audio solution that can run on a Raspberry Pi.  I suppose I should have added a link:

https://github.com/badaix/snapcast

mincequi, 2021-09-08:
@eriqjaffe 

Ok, i started working on this. It will take some more days.

mincequi, 2021-09-15:
@eriqjaffe 
Done. Please checkout the 0.8.0 release and edit the /etc/cornrowd.conf file accordingly.

Cornrowd will act as TCP client. So, you have to configure snapcast as TCP server.
Adding this line to your snapcast server should do the trick. Please notice the sample rate of 44100 (most Bluetooth sources will use 44.1k instead of 48k).
`source=tcp://127.0.0.1:4953?name=cornrow&sampleformat=44100:16:2`

Looking forward to your feedback.

eriqjaffe, 2021-09-16:
That works perfectly, thank you!

As a side note, I've run into errors where the snapcast service fails to start with a TCP entry referencing 127.0.0.1 in the .conf file.  I have found that using 0.0.0.0 instead of 127.0.0.1 works around that issue, so simply modifying both the snapserver.conf and the cornrowd.conf files appropriately does the job.

Thanks again, this is fantastic!

mincequi, 2021-09-16:
Good to hear, you're welcome. Btw. there is a donation option for cornrow 😉

Hmm, that's strange. For me "127.0.0.1" works on both sides. Maybe the ports had been used already by other streams?

closed 2021-11-26 by mincequi