mincequi/cornrow #20: Usage of non-SBC codecs

pdg, 2022-02-01:
Is it possible to use another Codec than SBC, e.g. aptx?

There is a codecs option in cornrowd.conf. But I found no information on other supported codecs and no other codec IDs in the [A2DB specification](https://www.bluetooth.com/de/specifications/specs/advanced-audi%e2%80%8b%e2%80%8bo-distribution-profile-1-3-2/). 

Thanks for the time you have invested in cornrowd. It works already great with SBC.

mincequi, 2022-02-02:
Yes, I had support for AAC and APTX in mind.

There are actually some resources available:
https://habr.com/en/post/456182/
https://github.com/Arkq/bluez-alsa/blob/master/src/a2dp-aptx.c

Hopefully, I find the time to integrate that. However, Mrs are always welcome :)

pdg, 2022-02-02:
> https://habr.com/en/post/456182/

That's an interesting article. Sticking with SBC in the meantime :-)
Would take a stab at e.g. aptx hd if I knew C++.

closed 2022-02-02 by pdg