mincequi/cornrow #21: Unmet dependency:  libboost-system1.67 on bullseye

infeeeee, 2022-04-12:
Thanks for this project, it seems nice!

I'm trying to install it from deb, on an Rpi 3B+, Raspbian 11 Bullseye, 5.10.63-v7+ kernel.

```shell
$ sudo apt install ./cornrowd_0.8.0_armhf.deb
[...]
The following packages have unmet dependencies:
 cornrowd : Depends: libboost-system1.67.0 but it is not installable
```

If I try to install manually `libboost-system` or `libboost-system-dev` it installs 1.74.

1.67 is not available in the bullseye repo, only 1.74: https://packages.debian.org/bullseye/libboost-system-dev

Should I build it manually, or can you update the build for bullseye? Thanks!

mincequi, 2022-04-13:
Ok, you are right. It should not depend on a specific version.  I will try to fix that within a couple of days.

Thanks for your feedback!

mincequi, 2022-05-02:
Fixed with v0.8.1

closed 2022-05-02 by mincequi