mincequi/cornrow #22: Compiling on Alpine

frafall, 2022-05-30:
I've compiled cornrow on Docker with an Alpine image, during this I found some issues addressed in the attached patch, also included the dockerfile builder. All tests have been on 0.8.1.

It basically boils down to a few issues

1. Use of "uint" in thirdparty/libcornrow/src/audio/Peq.cpp, assume size_t should be appropriate
2. Use of "uint" throughout cornrowd/src/config/Persistence.cpp, i typedef'ed it to "unsigned int"
3. Alpine uses an external lib for execinfo (not maintained and in disorder)
4. Use of __bswap_16 vs __bswap16 in cornrowd/src/config/Persistence.cpp
5. Unused variable in thirdparty/libcornrow/src/loguru/loguru.cpp(530)
 ```
char buff[256] __attribute__((unused));
 ```

7. Missing library in thirdparty/qtzeroprops/thirdparty/QtZeroConf/CMakeLists.txt, need avahi-common. This hit me in building (Docker) for debian bullseye as well.
 ```
    target_link_libraries(QtZeroConf PRIVATE avahi-client)
+  target_link_libraries(QtZeroConf PRIVATE avahi-common)
```

And that's about it, very nice and clean code :)

Now, the Alpine execinfo is in disorder and does core dump on calling backtrace.

-Morten

[alpine.zip](https://github.com/mincequi/cornrow/files/8798317/alpine.zip)

mincequi, 2022-05-30:
Thanks a lot for your feedback. This is very valuable.

I have one request for you. Could you open a regular pull request? This would help setting up a proper review, because i have some remarks :)

Thanks a lot!

-Manuel

closed 2022-10-26 by mincequi

amigthea, 2024-05-09:
> I've compiled cornrow on Docker with an Alpine image, during this I found some issues addressed in the attached patch, also included the dockerfile builder. All tests have been on 0.8.1.
> 
> It basically boils down to a few issues
> 
>     1. Use of "uint" in thirdparty/libcornrow/src/audio/Peq.cpp, assume size_t should be appropriate
> 
>     2. Use of "uint" throughout cornrowd/src/config/Persistence.cpp, i typedef'ed it to "unsigned int"
> 
>     3. Alpine uses an external lib for execinfo (not maintained and in disorder)
> 
>     4. Use of __bswap_16 vs __bswap16 in thirdparty/libcornrow/src/audio/AlsaSink.cpp
> 
>     5. Unused variable in thirdparty/libcornrow/src/loguru/loguru.cpp(530)
> 
> 
> ```
> char buff[256] __attribute__((unused));
> ```
> 
>     7. Missing library in thirdparty/qtzeroprops/thirdparty/QtZeroConf/CMakeLists.txt, need avahi-common. This hit me in building (Docker) for debian bullseye as well.
> 
> 
> ```
>    target_link_libraries(QtZeroConf PRIVATE avahi-client)
> +  target_link_libraries(QtZeroConf PRIVATE avahi-common)
> ```
> 
> And that's about it, very nice and clean code :)
> 
> Now, the Alpine execinfo is in disorder and does core dump on calling backtrace.
> 
> -Morten
> 
> [alpine.zip](https://github.com/mincequi/cornrow/files/8798317/alpine.zip)

can I ask you to share your docker alpine configuration? I was looking to build that too, did you had success with that? thank you

frafall, 2024-05-14:
> alpine.zip [1]
> Links:

I did not fully get this to work, but I've attached my patches and 
Dockerfile.
TBH it's been to long to I dont really remember why I put it to rest.

regards,
Morten

Den 2024-05-09 14:57, skrev amigthea: