mincequi/cornrow #24: Unable to Connect Bluetooth Device und Raspbian Buster and Bullseye

Snamhdaen, 2022-10-19:
I recently installed cornrow in order to set up a bluetooth integration for my snapserver instance. Snapserver runs on a Ubuntu 22 server and I have a raspberry which runs snapclient. I tried to install cornrow on this raspberry in order to enable bluetooth connectivity and send everything back to the snapserver via tcp. Installation was straightforward and Corocomo connects without issue.
However, when I try to actually connect a bluetooth device (e. g. Android phone) the connection is terminated after roughly 1 second. Journalctl registers the following error:

> bluetoothd[473]: Endpoint replied with an error: org.freedesktop.DBus.Error.NoReply
cornrowd[2567]: 20:16:54.698 [main thread     ]        Persistence.cpp:63    INFO| filter count: 0
cornrowd[2567]: 20:16:54.700 [main thread     ]       AudioManager.cpp:172   INFO| set transport> fd: -1, blocksize: 0, rate: 0
cornrowd[2567]: 20:16:54.700 [main thread     ]      TcpClientSink.cpp:206   INFO| TcpClientSink stopped
cornrowd[2567]: 20:16:54.700 [main thread     ]     SourceSelector.cpp:38    INFO| FdSource stopped

This is on raspbian buster (cornrow 0.8.0). I tried on raspbian bullseye (using cornrow 0.8.1) as well. The error was similar:

>bluetoothd[580]: profiles/audio/media.c:endpoint_reply() Endpoint replied with an error: org.freedesktop.DBus.Error.NoReply
bluetoothd[580]: profiles/audio/a2dp.c:select_cb() Unable to select a valid configuration
cornrowd[448]: 20:18:33.596 [main thread     ]        Persistence.cpp:63    INFO| filter count: 0
cornrowd[448]: 20:18:33.598 [main thread     ]       AudioManager.cpp:172   INFO| set transport> fd: -1, blocksize: 0, rate: 0
cornrowd[448]: 20:18:33.598 [main thread     ]      TcpClientSink.cpp:206   INFO| TcpClientSink stopped
cornrowd[448]: 20:18:33.598 [main thread     ]     SourceSelector.cpp:38    INFO| FdSource stopped

Bluetooth is configured correctly as far as I can tell since playback using pulseaudio (which I installed for a test) worked without altering bluetooth settings.

Thanks in advance and regards

mincequi, 2022-10-20:
Ok, this is strange. Can you show me the output of:
`sudo dbus-send --system --print-reply --type=method_call --dest='org.bluez' '/' org.freedesktop.DBus.ObjectManager.GetManagedObjects`

Search for `org.bluez.Media1`

Snamhdaen, 2022-10-21:
And it gets even stranger: For testing purposes I had installed pulseaudio and pulseaudio-bluetooth-modules. I removed both and reinstalled cornrow, unmasked and restarted the service. And it - worked... Just like that. Only to once again stop working after a reboot. Then it started throwing the errors I listed above.

Executing dbus-send and looking org.bluez.Media1 yields

> dict entry(
               string "org.bluez.Media1"
               array [
               ]
            )

I can provide the complete output if needed, but it's rather long.

mincequi, 2022-10-21:
Might be something with the access rights. Which pi do you exactly have?
This is an unmodified raspbian, correct?
Can you tell me your firmware version?

Snamhdaen, 2022-10-21:
Pi 3A+ running raspbian unmodified, kernel 5.15.61-v7+
Firmware:
Aug 26 2022 14:04:36
Copyright (c) 2012 Broadcom
version 102f1e848393c2112206fadffaaf86db04e98326 (clean) (release) (start)

Thanks for the swift reply!

mincequi, 2022-10-22:
The Pi3 is the only i have missing in my collection...
However, i can try out raspbian lite and check, if i have the same issues.

But anyhow, i doubt, that the issue is rooted in cornrow itself. It might be bluez-qt that causes trouble. Still wondering, why it runs once after installing/uninstalling pulse. Btw. i intentionally conflict with pulse, because i prefer a bare metal implementation instead of going via pulse.

I don't know, how skilled you are, but you can try checking out https://github.com/KDE/bluez-qt and see, whether there is a hint. Best would be, you install QtCreator (or another IDE) and debug yourself.

Did you also check different hardware? Another Pi, another phone?

Snamhdaen, 2022-10-23:
Yeah, I suspected that this issue is probably not a proper cornrow issue. What puzzles me, though, is that I started on a fresh unmodified install and can't get things to work.
I also checked different hardware (different pi, different Bluetooth device) with the same results.
Although debugging using QtCreator is probably beyond my capabilities, I currently suspect that you are correct about this issue having to do with access rights, specifically, in the DBus space. After removing Pulseaudio I didn't reboot. Some component must have lingered on and facilitated Bluetooth access for cornrow.
Perhaps this issue boils down to how the Bluetooth connection is established. I used the command line and bluetoothctl. Maybe this doesn't work for cornrow. Might I ask how you would recommend setting up Bluetooth? Is there a preferred method in order for cornrow to work? A systemwide service perhaps?

Many thanks!

mincequi, 2022-10-23:
Ok, i just installed raspbian light on a _Raspberry Pi 4 Model B Rev 1.1._
Firmware:
_Aug 26 2022 14:03:16 
Copyright (c) 2012 Broadcom
version 102f1e848393c2112206fadffaaf86db04e98326 (clean) (release) (start)_
OS:
_Linux raspberrypi 5.15.61-v7l+ #1579 SMP Fri Aug 26 11:13:03 BST 2022 armv7l GNU/Linux_

Then i installed `cornrowd` and fired it up as regular user. On my iPhone 12 my _raspberrypi_ appeared in bluetooth devices and then i simply connected to it. Music plays. Out of the box:

```
minski@raspberrypi:~ $ cornrowd 
time         [ thread name/id ]                   file:line     v| 
18:15:28.179 [main thread     ]           AlsaSink.cpp:240   INFO| Device stopped
Send property: QVariant(QUuid, QUuid("{ad10a100-d901-11e8-9f8b-f2801f1b9fd1}")) , value size: 0
Socket not connected
Send property: QVariant(QUuid, QUuid("{ad10e100-d901-11e8-9f8b-f2801f1b9fd1}")) , value size: 0
Socket not connected
TCP server published at port: 44737
18:16:43.021 [main thread     ]     AudioSinkAgent.cpp:64    INFO| Service 0000110D-0000-1000-8000-00805F9B34FB accepted from minskophon
18:16:43.022 [main thread     ]     AudioSinkAgent.cpp:64    INFO| Service 0000110E-0000-1000-8000-00805F9B34FB accepted from minskophon
onTransportVolumeChanged > transport volume: 127
Transport state: BluezQt::MediaTransport::State::Pending
operator() fd:  13 mtu read: 672 mtu write: 1004
18:17:00.874 [main thread     ]        Persistence.cpp:63    INFO| filter count: 0
18:17:00.875 [main thread     ]       AudioManager.cpp:172   INFO| set transport> fd: 14, blocksize: 672, rate: 44100
18:17:00.875 [main thread     ]         BufferPool.cpp:65    INFO| FdSource created new buffer. size: 672
Transport state: BluezQt::MediaTransport::State::Active
18:17:00.939 [main thread     ]     SourceSelector.cpp:52    INFO| Starting FdSource
18:17:00.939 [main thread     ]         RtpDecoder.cpp:67    INFO| sequence starts at: 1
18:17:00.940 [main thread     ]             Buffer.cpp:108   INFO| Buffer reallocated. 672 -> 3584 bytes
18:17:00.941 [main thread     ]             Buffer.cpp:108   INFO| Buffer reallocated. 3584 -> 8288 bytes
18:17:00.941 [main thread     ]           AlsaSink.cpp:240   INFO| Device stopped
18:17:00.962 [main thread     ]           AlsaSink.cpp:176   INFO| Device opened. delay: 100 ms, buffer: 500 ms
```

So, i am really out of ideas on how i could help you. :/

Snamhdaen, 2022-10-23:
So, I think it got it to work now. I followed in your footsteps, started again from a fresh install of raspbian lite and added cornrowd. Again nothing, connecting bluetooth devices was not possible. Then I added my user to bluetooth group via `usermod -a -G bluetooth pi` and suddenly it worked. Perhaps this is obvious, to me it wasn't. But I think access rights was the issue after all.

mincequi, 2022-10-23:
The installation should do this automatically for you:
```
minski@raspberrypi:~ $ sudo dpkg -i cornrowd_0.8.1_armhf.deb 
Selecting previously unselected package cornrowd.
(Reading database ... 44701 files and directories currently installed.)
Preparing to unpack cornrowd_0.8.1_armhf.deb ...
Unpacking cornrowd (0.8.1) ...
Setting up cornrowd (0.8.1) ...
Adding user `cornrowd' to group `audio' ...
Adding user cornrowd to group audio
Done.
Adding user `cornrowd' to group `bluetooth' ...
Adding user cornrowd to group bluetooth
Done.
Adding user `cornrowd' to group `video' ...
Adding user cornrowd to group video
Done.
minski@raspberrypi:~ $ 
```

Snamhdaen, 2022-10-24:
I mean I'm glad it's working now, but there is something really strange about it. I didn't explicitely start cornrowd with my user. But I guess it doesn't seem to be a cornrowd issue, so feel free to close this. Thanks a lot for your help. If I get to the bottom of this, I will report it here.

closed 2022-10-25 by mincequi