mincequi/cornrow #26: Issues building from source

TotalSpaceshipguy, 2023-02-01:
Hi

I'm trying to build the latest version from source on Debian (bullseye) ,on an amd64 processor which is fully updated.
All the listed dependencies have been installed, and I've got the latest source (git clone --recursive https://github.com/mincequi/cornrow).

The build process fails with this error...

Scanning dependencies of target cornrowd
[ 91%] Building CXX object cornrowd/CMakeFiles/cornrowd.dir/cornrowd_autogen/mocs_compilation.cpp.o
[ 91%] Building CXX object cornrowd/CMakeFiles/cornrowd.dir/src/Config.cpp.o
[ 92%] Building CXX object cornrowd/CMakeFiles/cornrowd.dir/src/Controller.cpp.o
[ 92%] Building CXX object cornrowd/CMakeFiles/cornrowd.dir/src/main.cpp.o
[ 92%] Linking CXX executable cornrowd
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/10/../../../x86_64-linux-gnu/libdns_sd.so: undefined reference to symbol 'avahi_string_list_free'
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/10/../../../x86_64-linux-gnu/libavahi-common.so: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [cornrowd/CMakeFiles/cornrowd.dir/build.make:182: cornrowd/cornrowd] Error 1
make[1]: *** [CMakeFiles/Makefile2:1117: cornrowd/CMakeFiles/cornrowd.dir/all] Error 2
make: *** [Makefile:149: all] Error 2

Any advice would be appreciated :)

mincequi, 2023-02-02:
This looks like an incompatible version of libavahi. Did you try
`sudo apt update && sudo apt upgrade`
?

I happen to have a fresh installation of debian bullseye on AMD64 and everything works perfect here...

TotalSpaceshipguy, 2023-02-03:
Hi

Thanks for the quick response :)
Yes, this was a fresh bullseye install and everything is up to date..These are the versions I have 

-rw-r--r-- 1 root root 130896 Aug  8 09:27 /usr/lib/x86_64-linux-gnu/libavahi-client.a
lrwxrwxrwx 1 root root     24 Aug  8 09:27 /usr/lib/x86_64-linux-gnu/libavahi-client.so -> libavahi-client.so.3.2.9
lrwxrwxrwx 1 root root     24 Aug  8 09:27 /usr/lib/x86_64-linux-gnu/libavahi-client.so.3 -> libavahi-client.so.3.2.9
-rw-r--r-- 1 root root  75952 Aug  8 09:27 /usr/lib/x86_64-linux-gnu/libavahi-client.so.3.2.9
**-rw-r--r-- 1 root root  93722 Aug  8 09:27 /usr/lib/x86_64-linux-gnu/libavahi-common.a
lrwxrwxrwx 1 root root     24 Aug  8 09:27 /usr/lib/x86_64-linux-gnu/libavahi-common.so -> libavahi-common.so.3.5.4
lrwxrwxrwx 1 root root     24 Aug  8 09:27 /usr/lib/x86_64-linux-gnu/libavahi-common.so.3 -> libavahi-common.so.3.5.4
-rw-r--r-- 1 root root  51496 Aug  8 09:27 /usr/lib/x86_64-linux-gnu/libavahi-common.so.3.5.4**
-rw-r--r-- 1 root root 494968 Aug  8 09:27 /usr/lib/x86_64-linux-gnu/libavahi-core.a
lrwxrwxrwx 1 root root     22 Aug  8 09:27 /usr/lib/x86_64-linux-gnu/libavahi-core.so -> libavahi-core.so.7.1.0
lrwxrwxrwx 1 root root     22 Aug  8 09:27 /usr/lib/x86_64-linux-gnu/libavahi-core.so.7 -> libavahi-core.so.7.1.0
-rw-r--r-- 1 root root 231336 Aug  8 09:27 /usr/lib/x86_64-linux-gnu/libavahi-core.so.7.1.0

So I have package v3.5.4 which is based on the avahi 0.8 source. This looks correct according to the file in the debian package list - https://packages.debian.org/bullseye/amd64/libavahi-common3/filelist

I have do have another debian and ubuntu, install and they both have the same file version and sizes as the build environment

The build environment is actually a debian bullseye lxc container running on ubuntu 22.04 - maybe that's upsetting things ? I'll spin up a  fresh virtualbox debian VM and try again

TotalSpaceshipguy, 2023-02-03:
OK. I've created a brand new debian vm using the latest 11-6 netinst ISO
Ran an apt update and apt upgrade.. Installed all the listed dependencies and tried to build the source again.

I still get the same build error unfortunately :(

TotalSpaceshipguy, 2023-02-07:
Hi @mincequi 

I've managed to fix my build issue :) From what I can see, the thirdparty/qtzeroprops/thirdparty/QtZeroConf/CMakeLists.txt file in your master branch doesn't include the avahi-common library

I was able to fix the build by fixed my issue changing line 34 of that file from
    target_link_libraries(QtZeroConf PRIVATE avahi-client)
to
    target_link_libraries(QtZeroConf PRIVATE avahi-client avahi-common)

The current master of QtZeroConf  at https://github.com/jbagg/QtZeroConf/blob/master/CMakeLists.txt seems to be a later one than your fork, and that version does also include a reference to avahi-common 

I'm not sure why your build works. Maybe you have different source locally that hasn't been added to the github master yet, or your local setup is using a more up to date version of the QtZeroConf project, but I could not get the build to work by cloning the source into a new folder.

I'd love to pull request, but wouldn't know where to start lol - I haven't done any development for over 10 years ! 

Anyway, hope this helps someone else. An official amd64 release of the 0.8.1 deb would be handy as I couldn't see one, which lead me down this rabbit hole, but then I wouldn't have learnt anything ;)

Cheers

mincequi, 2023-02-07:
Hi @TotalSpaceshipguy 

great to hear, that you had been able to fix that :)

Thanks for your input. I will update my refs to the most recent QtZeroConf.

Ok, just discovered that i only have armhf builds since this is my main target. However, you are absolutely right. Providing binaries for amd64 would ease up things...

...when i find the time, this project needs some cleaning anyway. ;)

Regards

carnei-ro, 2023-04-12:
@TotalSpaceshipguy thank for your tip about line 34 of  `thirdparty/qtzeroprops/thirdparty/QtZeroConf/CMakeLists.txt`.

carnei-ro, 2023-04-12:
hey guys, cornrowd is conflicting with pulseaudio:

```
$ sudo apt install ./cornrowd_0.8.1_arm64.deb 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'cornrowd' instead of './cornrowd_0.8.1_arm64.deb'
The following packages were automatically installed and are no longer required:
  libltdl7 libpulsedsp libspeexdsp1 libtdb1 libwebrtc-audio-processing1 pulseaudio-utils
Use 'sudo apt autoremove' to remove them.
The following packages will be REMOVED:
  pulseaudio
The following NEW packages will be installed:
  cornrowd
```

do you know why?

mincequi, 2023-04-12:
Hey @carnei-ro,
yes, this is intended. cornrowd aims to be a "bare-metal" bluetooth daemon for embedded devices. So, I do not support output via pulseaudio since pulse is very hardware hungry and supports bluetooth itself.
However, I often thought about to change that, but I ran completely out of time...

closed 2023-10-27 by mincequi