mincequi/cornrow #28: No audio in snapcast

ea5522, 2023-07-22:
Hello. i was able to get bluetooth working. Laptop or cellphone connects but there is no audio going into snapcast

cornrow config:
[tcp_sink]
host = 127.0.0.1
port = 4954

snapcast config:
source = tcp://127.0.0.1:4954?name=Bluetooth&mode=server&sampleformat=44100:16:2

root@snapcast:/home/snapcast# journalctl -b | grep bluetooth
Jul 22 19:54:32 snapcast bluetoothd[802]: Bluetooth daemon 5.64
Jul 22 19:54:32 snapcast bluetoothd[802]: Starting SDP server
Jul 22 19:54:32 snapcast bluetoothd[802]: Bluetooth management interface 1.21 initialized
Jul 22 19:54:33 snapcast bluetoothd[802]: Endpoint registered: sender=:1.16 path=/MediaEndpoint/Sink/Sbc
Jul 22 19:54:33 snapcast cornrowd[913]: Waiting for bluetooth audio source to connect. Ctrl + C to cancel...
Jul 22 19:55:12 snapcast bluetoothd[802]: /org/bluez/hci0/dev_(REDACTED)/fd0: fd(29) ready

root@snapcast:/home/snapcast# journalctl -b | grep cornrowd
Jul 22 19:54:33 snapcast cornrowd[913]: Waiting for bluetooth audio source to connect. Ctrl + C to cancel...
Jul 22 19:55:10 snapcast cornrowd[913]: Accepted service: "(REDACTED)" from "(REDACTED)"
Jul 22 19:55:11 snapcast cornrowd[913]: Accepted service: "(REDACTED)" from "(REDACTED)"
Jul 22 19:55:11 snapcast cornrowd[913]: sampleRate:  44100
Jul 22 19:55:12 snapcast cornrowd[913]: Transport state: BluezQt::MediaTransport::State::Pending
Jul 22 19:55:12 snapcast cornrowd[913]: operator() fd:  8 mtu read: 672 mtu write: 672
Jul 22 19:55:12 snapcast cornrowd[913]: 19:55:12.051 FileDescriptorSource.cp:25    INFO| opening file descriptor: 9
Jul 22 19:55:12 snapcast cornrowd[913]: Transport state: BluezQt::MediaTransport::State::Active
Jul 22 19:55:12 snapcast cornrowd[913]: 19:55:12.350 FileDescriptorSource.cp:54    WARN| cannot estimate number of slices
Jul 22 19:55:12 snapcast cornrowd[913]: 19:55:12.351          SbcDecoder.cpp:73    WARN| Sync byte incorrect
Jul 22 19:55:12 snapcast cornrowd[913]: 19:55:12.351 FileDescriptorSource.cp:54    WARN| cannot estimate number of slices
Jul 22 19:55:12 snapcast cornrowd[913]: 19:55:12.351          SbcDecoder.cpp:42    WARN| RTP header invalid

What could tbe the issue?

mincequi, 2023-08-07:
Which version are you actually using? This seems like an older version.
Apparently, the bluetooth stream seams corrupted somehow. Which kind of computer are you actually using?

ea5522, 2023-08-08:
It is an intel i5, 8gb ram lenovo tiny running esxi 6.5 or 6.7(have to check) . I have this on a vm running ubuntu

mincequi, 2023-10-27:
@ea5522 
Oh, running this within a VM might be troublesome. Ever tried in a native environment?

closed 2023-12-17 by mincequi