mincequi/cornrow #29: unable to start cornrowd.service

iamhai, 2023-10-27:
Hi mincequi,

I have built and installed cornrow package, but it's not possible for me to start cornrowd.service. Please help.

My device:
`Linux armbian 6.1.55-ophub #1 SMP Sat Sep 23 07:37:26 EDT 2023 aarch64 aarch64 aarch64 GNU/Linux`
```
~#systemctl start cornrowd.service
Failed to start cornrowd.service: Unit avahi-daemon.service not found.
```

```
~#cornrowd
time         [ thread name/id ]                   file:line     v|
cornrowd: entrygroup.c:108: avahi_entry_group_new: Assertion "client" failed.

Loguru caught a signal: SIGABRT
Stack trace:
11        0x558e75abb0 cornrowd(+0x2abb0) [0x558e75abb0]
10        0x7f925474cc __libc_start_main + 152
9         0x7f925473fc /lib/aarch64-linux-gnu/libc.so.6(+0x273fc) [0x7f925473fc]
8         0x558e757768 cornrowd(+0x27768) [0x558e757768]
7         0x558e781d4c cornrowd(+0x51d4c) [0x558e781d4c]
6         0x7f93095e40 avahi_entry_group_commit + 0
5         0x7f92554040 __assert_perror_fail + 0
4         0x7f92553fd0 /lib/aarch64-linux-gnu/libc.so.6(+0x33fd0) [0x7f92553fd0]
3         0x7f92547130 abort + 228
2         0x7f9255a67c raise + 28
1         0x7f9259f200 /lib/aarch64-linux-gnu/libc.so.6(+0x7f200) [0x7f9259f200]
0         0x7f94d2e7ac __kernel_rt_sigreturn + 0
17:44:50.666 [main thread     ]                       :0     FATL| Signal: SIGABRT
Aborted
```

```
~#journalctl -b | grep cornrowd
Oct 27 16:55:44 armbian groupadd[261390]: group added to /etc/group: name=cornrowd, GID=121
Oct 27 16:55:44 armbian groupadd[261390]: group added to /etc/gshadow: name=cornrowd
Oct 27 16:55:44 armbian groupadd[261390]: new group: name=cornrowd, GID=121
Oct 27 16:55:44 armbian useradd[261399]: new user: name=cornrowd, UID=115, GID=121, home=/var/lib/cornrowd, shell=/usr/sbin/nologin, from=/dev/pts/2
Oct 27 16:55:45 armbian usermod[261407]: change user 'cornrowd' password
Oct 27 16:55:45 armbian chage[261417]: changed password expiry for cornrowd
Oct 27 16:55:45 armbian gpasswd[261430]: user cornrowd added by root to group audio
Oct 27 16:55:45 armbian gpasswd[261445]: user cornrowd added by root to group bluetooth
Oct 27 16:55:45 armbian gpasswd[261460]: user cornrowd added by root to group video
```

mincequi, 2023-10-27:
Hi @iamhai,
it seems, that the avahi-daemon is not running on your machine.
Can you try installing it?

`sudo apt-get install avahi-daemon `

iamhai, 2023-10-27:
Thank you for very quick response. I'll try it tomorrow and let you know.

iamhai, 2023-10-28:
After installing avahi-daemon, I'm able to start the service. But now I have another problem:

cornrow config:
```
# This is a TOML document.

[bluetooth_source]
codecs = [ 0x0 ]        # As defined in the A2DP specification (SBC (0x0) is mandatory)
sample_rates = [ 44100, 48000 ]

#[airplay_source]
#name = "myAirplay"
# port = 0              # port to listen on (0 for auto select)
# buffer_time = 2000    # in ms (defaults to 2000ms)

#[alsa_sink]
#device = "default"

[tcp_sink]
host = "127.0.0.1"
port = 4953
```
snapcast config:
`tcp://127.0.0.1:4953?name=Bluetooth&mode=server&sampleformat=44100:16:2`

but cornrows server published at port: 10927

```
Oct 28 17:33:11 armbian systemd[1]: Started Cornrow daemon.
Oct 28 17:33:12 armbian cornrowd[1022]: time         [ thread name/id ]                   file:line     v|
Oct 28 17:33:12 armbian cornrowd[1022]: 17:33:12.869 [main thread     ]      TcpClientSink.cpp:206   INFO| TcpClientSink stopped
Oct 28 17:33:12 armbian cornrowd[1022]: Send property: QVariant(QUuid, QUuid("{ad10e100-d901-11e8-9f8b-f2801f1b9fd1}")) , value size: 0
Oct 28 17:33:12 armbian cornrowd[1022]: Socket not connected
Oct 28 17:33:12 armbian cornrowd[1022]: Send property: QVariant(QUuid, QUuid("{ad10a100-d901-11e8-9f8b-f2801f1b9fd1}")) , value size: 0
Oct 28 17:33:12 armbian cornrowd[1022]: Socket not connected
Oct 28 17:33:13 armbian cornrowd[1022]: TCP server published at port: 10927
Oct 28 17:41:51 armbian cornrowd[1022]: 17:41:51.954 [main thread     ]     AudioSinkAgent.cpp:64    INFO| Service 0000110D-0000-1000-8000-00805F9B34FB accepted from Pixel 6a
Oct 28 17:41:59 armbian cornrowd[1022]: Transport state: BluezQt::MediaTransport::State::Pending
Oct 28 17:41:59 armbian cornrowd[1022]: operator() fd:  13 mtu read: 672 mtu write: 1024
```

when start playing audio from my phone, I got this:
```
Oct 28 17:41:59 armbian cornrowd[1022]: 17:41:59.876 [main thread     ]        Persistence.cpp:63    INFO| filter count: 0
Oct 28 17:41:59 armbian cornrowd[1022]: 17:41:59.878 [main thread     ]       AudioManager.cpp:172   INFO| set transport> fd: 14, blocksize: 672, rate: 44100
Oct 28 17:41:59 armbian cornrowd[1022]: 17:41:59.878 [main thread     ]         BufferPool.cpp:65    INFO| FdSource created new buffer. size: 672
Oct 28 17:41:59 armbian cornrowd[1022]: Transport state: BluezQt::MediaTransport::State::Active
Oct 28 17:41:59 armbian cornrowd[1022]: 17:41:59.963 [main thread     ]     SourceSelector.cpp:52    INFO| Starting FdSource
Oct 28 17:41:59 armbian cornrowd[1022]: 17:41:59.963 [main thread     ]      TcpClientSink.cpp:201   INFO| TcpClientSink started
Oct 28 17:41:59 armbian cornrowd[1022]: 17:41:59.964 [main thread     ]         RtpDecoder.cpp:67    INFO| sequence starts at: 1
Oct 28 17:41:59 armbian cornrowd[1022]: 17:41:59.964 [main thread     ]             Buffer.cpp:108   INFO| Buffer reallocated. 672 -> 3584 bytes
Oct 28 17:41:59 armbian cornrowd[1022]: 17:41:59.964 [main thread     ]             Buffer.cpp:108   INFO| Buffer reallocated. 3584 -> 8288 bytes
Oct 28 17:41:59 armbian cornrowd[1022]: 17:41:59.965 [main thread     ]      TcpClientSink.cpp:220   WARN| Connection refused
Oct 28 17:41:59 armbian cornrowd[1022]: 17:41:59.965 [main thread     ]      TcpClientSink.cpp:223   INFO| 0 bytes of 2560 written
Oct 28 17:41:59 armbian cornrowd[1022]: 17:41:59.965 [main thread     ]      TcpClientSink.cpp:122   WARN| Error: Operation canceled when connecting endpoint: 0.0.0.0:0
Oct 28 17:41:59 armbian cornrowd[1022]: 17:41:59.993 [main thread     ]      TcpClientSink.cpp:105   WARN| Connection refused when connecting host: 127.0.0.1
Oct 28 17:42:00 armbian cornrowd[1022]: 17:42:00.022 [main thread     ]      TcpClientSink.cpp:105   WARN| Connection refused when connecting host: 127.0.0.1
Oct 28 17:42:00 armbian cornrowd[1022]: 17:42:00.051 [main thread     ]      TcpClientSink.cpp:105   WARN| Connection refused when connecting host: 127.0.0.1
Oct 28 17:42:00 armbian cornrowd[1022]: 17:42:00.107 [main thread     ]      TcpClientSink.cpp:105   WARN| Connection refused when connecting host: 127.0.0.1
```
So I reconfig and restart snapcast with:
`tcp://127.0.0.1:10927?name=Bluetooth&mode=server&sampleformat=44100:16:2`
but still got the same problem: 
`TcpClientSink.cpp:105   WARN| Connection refused when connecting host: 127.0.0.1`

mincequi, 2023-10-30:
Hi @iamhai,
> but cornrows server published at port: 10927
Don't get irritated by this. Cornrow opens another TCP port for remote controlling the internal equalizer.

But this connection refused error is strange. Can you verify, that the Snapcast host (and port) are really open?
Using nmap or telnet?

iamhai, 2023-10-31:
@mincequi ,
I've checked the port 4953, but it seems not to open:
```
root@armbian:~# telnet 192.168.1.110 4953
Trying 192.168.1.110...
telnet: Unable to connect to remote host: Connection refused
```

```
root@armbian:~# nmap 192.168.1.110
Starting Nmap 7.80 ( https://nmap.org ) at 2023-10-31 08:38 +07
Nmap scan report for 192.168.1.110
Host is up (0.000042s latency).
Not shown: 998 closed ports
PORT    STATE SERVICE
22/tcp  open  ssh
111/tcp open  rpcbind

Nmap done: 1 IP address (1 host up) scanned in 0.32 seconds
```

```
root@armbian:~# netstat -lntu
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State
tcp        0      0 0.0.0.0:29321           0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:6680            0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:111             0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:12635           0.0.0.0:*               LISTEN
tcp        0      0 192.168.1.110:40001     0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:1704            0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:1705            0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:1780            0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:8123            0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:6052            0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:3179            0.0.0.0:*               LISTEN
tcp        0      0 127.0.0.53:53           0.0.0.0:*               LISTEN
tcp6       0      0 :::6680                 :::*                    LISTEN
tcp6       0      0 :::22                   :::*                    LISTEN
tcp6       0      0 :::111                  :::*                    LISTEN
tcp6       0      0 :::8123                 :::*                    LISTEN
tcp6       0      0 fe80::ad0d:8d77:e:40000 :::*                    LISTEN
udp        0      0 0.0.0.0:50871           0.0.0.0:*
udp        0      0 0.0.0.0:61231           0.0.0.0:*
udp        0      0 0.0.0.0:1900            0.0.0.0:*
udp        0      0 0.0.0.0:1900            0.0.0.0:*
udp        0      0 127.0.0.53:53           0.0.0.0:*
udp        0      0 0.0.0.0:61499           0.0.0.0:*
udp        0      0 0.0.0.0:111             0.0.0.0:*
udp        0      0 127.0.0.1:323           0.0.0.0:*
udp        0      0 0.0.0.0:12752           0.0.0.0:*
udp        0      0 0.0.0.0:5353            0.0.0.0:*
udp        0      0 192.168.1.110:5353      0.0.0.0:*
udp        0      0 0.0.0.0:5353            0.0.0.0:*
udp        0      0 172.17.0.1:5353         0.0.0.0:*
udp        0      0 127.0.0.1:5353          0.0.0.0:*
udp        0      0 192.168.1.110:5353      0.0.0.0:*
udp        0      0 0.0.0.0:5353            0.0.0.0:*
udp        0      0 0.0.0.0:5353            0.0.0.0:*
udp6       0      0 :::36492                :::*
udp6       0      0 :::65284                :::*
udp6       0      0 :::1900                 :::*
udp6       0      0 :::1900                 :::*
udp6       0      0 :::111                  :::*
udp6       0      0 ::1:323                 :::*
udp6       0      0 :::27159                :::*
udp6       0      0 fe80::ad0d:8d77:e2d:546 :::*
udp6       0      0 :::62136                :::*
udp6       0      0 :::5353                 :::*
udp6       0      0 :::5353                 :::*
udp6       0      0 :::5353                 :::*
udp6       0      0 fe80::ad0d:8d77:e2:5353 :::*
udp6       0      0 :::5353                 :::*
```
ufw is not installed

snapserver is running:
```
root@armbian:~# systemctl status snapserver
● snapserver.service - Snapcast server
     Loaded: loaded (/lib/systemd/system/snapserver.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2023-10-31 08:25:20 +07; 23min ago
       Docs: man:snapserver(1)
   Main PID: 3713 (snapserver)
      Tasks: 7 (limit: 4210)
     Memory: 2.5M
        CPU: 14.942s
     CGroup: /system.slice/snapserver.service
             ├─3713 /usr/bin/snapserver --logging.sink=system --server.datadir=/var/lib/snapserver

```

mincequi, 2023-10-31:
Hi @iamhai,

> I've checked the port 4953, but it seems not to open

Well, that sounds like a clear sign, that snapcast's port is not open for accepting streams from cornrow. Please check your snapcast config (i am not an expert on this. i just added this feature, because multiple users requested it).

iamhai, 2023-11-01:
Hi @mincequi ,
you won't imagine how silly I'm and how simple the solution was:
In your doc, you wrote:
```
And this to your snapcast configuration:
tcp://<listen IP, e.g. 127.0.0.1>:<port>?name=<name>[&mode=server]&sampleformat=44100:16:2
```
so I insert that line without putting `source = ` at the beginning of the line.
Now it connected to my phone, but when try to play some music I receive only some intermittent sound. Playing from mopidy via tcp works well.

mincequi, 2023-11-05:
Hi @iamhai,
do you play via bluetooth or airplay? Did you check the section about raspberry pi issues? Are you using a raspberry pi?
If you provide some more details about your environment my support could be much faster and detailed ;)

closed 2023-12-17 by mincequi