mincequi/cornrow #30: terminate called after throwing an instance of 'std::out_of_range'

detroitgoose, 2023-10-30:
Hello, and thank you for cornrow! When it works it's works so well...

I'm using this on a raspberry 4b with Raspbian GNU/Linux 11 (bullseye) and streamed as a snapcast source.

The following shows the status of the service after it has been aborted. If I restart the service it works well 

> ● cornrowd.service - Cornrow daemon
     Loaded: loaded (/lib/systemd/system/cornrowd.service; enabled; vendor preset: enabled)
     Active: failed (Result: signal) since Sat 2023-10-28 10:57:37 EDT; 2 days ago
    Process: 14018 ExecStart=/usr/bin/cornrowd (code=killed, signal=ABRT)
   Main PID: 14018 (code=killed, signal=ABRT)
        CPU: 56min 58.912s

> Oct 28 10:57:37 hifikitchen cornrowd[14018]: terminate called after throwing an instance of 'std::out_of_range'
Oct 28 10:57:37 hifikitchen cornrowd[14018]:   what():  map::at
Oct 28 10:57:37 hifikitchen cornrowd[14018]: Loguru caught a signal: SIGABRT
Oct 28 10:57:37 hifikitchen cornrowd[14018]: Stack trace:
Oct 28 10:57:37 hifikitchen cornrowd[14018]: 1   0xf53f1420 gsignal + 188
Oct 28 10:57:37 hifikitchen cornrowd[14018]: 0   0xf53f2910 __default_rt_sa_restorer + 0
Oct 28 10:57:37 hifikitchen cornrowd[14018]: 10:57:37.211 [main thread     ]                       :0     FATL| Signal: SIGABRT
Oct 28 10:57:37 hifikitchen systemd[1]: cornrowd.service: Main process exited, code=killed, status=6/ABRT
Oct 28 10:57:37 hifikitchen systemd[1]: cornrowd.service: Failed with result 'signal'.
Oct 28 10:57:37 hifikitchen systemd[1]: cornrowd.service: Consumed 56min 58.912s CPU time.

And here is the output after a system reboot:

> cornrowd.service - Cornrow daemon
     Loaded: loaded (/lib/systemd/system/cornrowd.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2023-10-30 12:26:46 EDT; 3min 43s ago
   Main PID: 462 (cornrowd)
      Tasks: 3 (limit: 2137)
        CPU: 15.791s
     CGroup: /system.slice/cornrowd.service
             └─462 /usr/bin/cornrowd

> Oct 30 12:26:55 hifikitchen cornrowd[462]: *** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
Oct 30 12:26:55 hifikitchen cornrowd[462]: *** WARNING *** Please fix your application to use the native API of Avahi!
Oct 30 12:26:55 hifikitchen cornrowd[462]: *** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
Oct 30 12:26:55 hifikitchen cornrowd[462]: 12:26:55.871 [main thread     ]AirplaySourcePrivate.cp:66    INFO| reception started. name: KitchenAirplay, rt>
Oct 30 12:26:55 hifikitchen cornrowd[462]: 12:26:55.871 [main thread     ]      TcpClientSink.cpp:206   INFO| TcpClientSink stopped
Oct 30 12:26:55 hifikitchen cornrowd[462]: Send property: QVariant(QUuid, QUuid("{ad10e100-d901-11e8-9f8b-f2801f1b9fd1}")) , value size: 0
Oct 30 12:26:55 hifikitchen cornrowd[462]: Socket not connected
Oct 30 12:26:55 hifikitchen cornrowd[462]: Send property: QVariant(QUuid, QUuid("{ad10a100-d901-11e8-9f8b-f2801f1b9fd1}")) , value size: 0
Oct 30 12:26:55 hifikitchen cornrowd[462]: Socket not connected
Oct 30 12:26:56 hifikitchen cornrowd[462]: TCP server published at port: 36539

I have 3 sources in my snapcast config: cornrow, mopidy, and pianobar (pandora). I can easily switch sources using home assistant and nodered and everything works as expected until the abort service error. I've tried connecting both airplay and BT at the same time, and while AP took precedence, the BT connection took over once AP was disconnected.

I am not currently able to define when the problem happens. If I can provide further detail please advise.

Cheers

detroitgoose, 2023-10-30:
Adding cornrow config if pertinent:

> [bluetooth_source]
codecs = [ 0x0 ]        # As defined in the A2DP specification (SBC (0x0) is mandatory)
sample_rates = [ 44100, 48000 ]

> [airplay_source]
name = "KitchenAirplay"
 port = 0              # port to listen on (0 for auto select)
 buffer_time = 2000    # in ms (defaults to 2000ms)

> #[alsa_sink]
#device = "default"

> [tcp_sink]`
host = "127.0.0.1"
port = 4953

mincequi, 2023-10-30:
Hmm, this map out-of-range is suspicious.
But in any case: your config file has an error: [tcp_sink]`<- this symbol might cause an error when parsing the config file.

Maybe you can produce a stack trace?

detroitgoose, 2023-10-30:
@mincequi

> your config file has an error: [tcp_sink]`<- this symbol might cause an error when parsing the config file.

Oops, that was a result of my poor github markup. I've edited the post to show the config as it actually is: without the backtick.

> Maybe you can produce a stack trace?

I tried `sudo strace -s 99 -ffp 766`, 776 being the process ID, but I'm not sure how helpful the data is. I get the following response which repeats ad infinitum:

```
[pid   766] clock_gettime64(CLOCK_MONOTONIC, {tv_sec=15991, tv_nsec=591824800}) = 0
[pid   766] clock_gettime64(CLOCK_MONOTONIC, {tv_sec=15991, tv_nsec=591944428}) = 0
[pid   766] epoll_wait(10, [], 128, 0)  = 0
[pid   766] clock_gettime64(CLOCK_MONOTONIC, {tv_sec=15991, tv_nsec=592170463}) = 0
[pid   766] poll([{fd=3, events=POLLIN}, {fd=6, events=POLLIN}, {fd=7, events=POLLIN}, {fd=8, events=POLLIN}], 4, 1) = 0 (Timeout)
```
This was done while the program is running without fault.

If you could tell me the proper way to do a stack trace, I'll run it next time the error occurs.

mincequi, 2023-10-31:
@detroitgoose,

Ok, having a stack trace would not help, if the tools runs without fault. It depends a little bit on your environment, but checkout some documentation on how to enable "core dumps" (sorry, the wording stack trace might have been misleading).
https://wiki.archlinux.org/title/Core_dump

If the tool crashes, you get a core file. Together with cornrow's executable you can trace back the issue:
gdb <executable> <core-file>

detroitgoose, 2023-11-12:
@mincequi

I enabled core dumps 2 weeks ago, but it hasn't crashed since.

Not sure what happened but I'll mark this closed. Thanks again for such seamless BT/AP integration

closed 2023-11-12 by detroitgoose