mincequi/cornrow #31: Problems with sound

Ithu123, 2024-01-07:
Hi, first of all thank you for cornrow! 

A few things (let me know if i should open more than one issue)

1. At least on my rpi 4 the install via apt is broken due to some dependencies (libavcodec58 i think), which are outdated.
2. When connecting via cronrow to my rpi4 i get an output (via snapcast), but its pitched higher and stuttering. I tried disableing BluetoothLE, but the problem remains.
3. I tried installing the EQ App to investigate if the EQ causes the problem with the higher pitch, but the app is not available for Android anymore. Is there a way to disable EQ completely in the conf?
4. Are there other codecs implementet? 

Thank you in advance.

Ithu123, 2024-01-07:
Okay just a few seconds later i realize im an idiot and the samplerate was wrong in my Snapcast conf. Im not editing it though, because maybe someone else experiences the same problem and finds this. The other two points remain valid.

mincequi, 2024-01-08:
Hey @Ithu123 ,
nice to hear, that people are still interested in my little tool and thanks for your feedback.
Answering your points:
> 1. At least on my rpi 4 the install via apt is broken due to some dependencies (libavcodec58 i think), which are outdated.
Yes, this might be the case, since libavcodec has fixed and exact versions which i need to link to. We need to recompile and use newer version of that.
> 3. I tried installing the EQ App to investigate if the EQ causes the problem with the higher pitch, but the app is not available for Android anymore. Is there a way to disable EQ completely in the conf?
The EQ won't be used if no filter is set. So, it is disabled by default.
> 4. Are there other codecs implementet?
No, i only implemented SBC, because i believe it is fairly good enough in terms of quality. Check issue #20 for further reading.

My current problem in maintaining this is, that i am running completely out of time and working on other things, atm. But if someone can provide PRs, i am willing to integrate :)

closed 2024-03-02 by mincequi