mincequi/cornrow #32: Debian bookworm support?

jreusch, 2024-03-10:
hey hey
great project, thanks for sharing it.
it worked as bluetooth to snapcast solution like a charm, but after upgrading to debian 12 based raspbian it no longer does :/

>  /usr/bin/cornrowd: error while loading shared libraries: libmmal_core.so.0: cannot open shared object file: No such file or directory

i'm currently searching around but can only find some dropped mmal support e.g. here
https://github.com/raspberrypi/userland/issues/688

also the paths in /opt/vc/ where it lied up until bullseye is gone.

currently trying to recompile it and see if it works, but also fighting, so i though maybe already somebody has hit this problem?
thanks!

jreusch, 2024-03-10:
ok, unsurprisingly compiling on my own did the trick.

first checked out git and used main branch which failed, but using the released version wokred nicely (though compilation aborted three times, but in the end there was a nice debian package)
nice, thanks a lot again for your work!

closed 2024-03-10 by jreusch

mincequi, 2024-03-10:
Hi @jreusch,
thanks for your feedback. Currently, i do not have a proper bookworm environment, but i could already find several issues with newer library versions. I might find the time to fix this on master in a couple of weeks.