mincequi/cornrow #33: Connection problems with smartphone

Blaubeeree, 2024-04-19:
I've installed cornrow v0.9.0 with the deb file on my raspberry 4, but when I try connecting to it with bluetooth on my android smartphone I get constant audio stuttering or can't connect at all.

When the connection fails cornrow doesn't give me any logs. When it connects it shows something about a "broken pipe" the whole time, I would send the specific log message but I currently can't get my smartphone to connect.

Is there anything I can do to fix this, I never had problems with bluetooth headphones or similar.

I also tried connecting with my PC, that works without problems.

mincequi, 2024-04-20:
Can you try an older version of cornrow? Does another bluetooth implementation work for your phone?
What do you mean with connecting with your PC? Can your PC send audio to cornrow?

Blaubeeree, 2024-04-20:
> Can you try an older version of cornrow?

I'll try later

> Does another bluetooth implementation work for your phone?

Do you mean connection with another bluetooth software then cornrow to my raspberry? Then idk, haven't tried one

Or do you mean connecting some other device to my smartphone? Then yes

> What do you mean with connecting with your PC? Can your PC send audio to cornrow?

Yes my PC connected to cornrow with bluetooth without problems and can send audio

Blaubeeree, 2024-04-20:
> Can you try an older version of cornrow?

Same issue with audio stuttering on 0.8.1 0.8.0 and 0.7.0

But it connected again so I can give you a few logs:
```
21:06:59.414 [main thread     ]     AudioSinkAgent.cpp:64    INFO| Service 0000110D-0000-1000-8000-00805F9B34FB accepted from Galaxy A52s
Transport state: BluezQt::MediaTransport::State::Pending
operator() fd:  13 mtu read: 672 mtu write: 1024
21:07:19.188 [main thread     ]        Persistence.cpp:63    INFO| filter count: 0
21:07:19.189 [main thread     ]       AudioManager.cpp:172   INFO| set transport> fd: 14, blocksize: 672, rate: 44100
21:07:19.190 [main thread     ]         BufferPool.cpp:65    INFO| FdSource created new buffer. size: 672
Transport state: BluezQt::MediaTransport::State::Active
21:07:19.292 [main thread     ]     SourceSelector.cpp:52    INFO| Starting FdSource
21:07:19.292 [main thread     ]         RtpDecoder.cpp:67    INFO| sequence starts at: 1
21:07:19.293 [main thread     ]             Buffer.cpp:108   INFO| Buffer reallocated. 672 -> 2872 bytes
21:07:19.293 [main thread     ]             Buffer.cpp:108   INFO| Buffer reallocated. 2872 -> 6204 bytes
21:07:19.293 [main thread     ]           AlsaSink.cpp:240   INFO| Device stopped
21:07:19.307 [main thread     ]           AlsaSink.cpp:176   INFO| Device opened. delay: 100 ms, buffer: 500 ms
21:07:20.584 [main thread     ]           AlsaSink.cpp:281   WARN| Write failed: Broken pipe
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
21:07:21.986 [main thread     ]           AlsaSink.cpp:281   WARN| Write failed: Broken pipe
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
21:07:22.837 [main thread     ]           AlsaSink.cpp:281   WARN| Write failed: Broken pipe
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
21:07:24.344 [main thread     ]           AlsaSink.cpp:281   WARN| Write failed: Broken pipe
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
Transport state: BluezQt::MediaTransport::State::Idle
21:07:24.748 [main thread     ]        Persistence.cpp:63    INFO| filter count: 0
21:07:24.749 [main thread     ]       AudioManager.cpp:172   INFO| set transport> fd: -1, blocksize: 0, rate: 44100
21:07:24.794 [main thread     ]           AlsaSink.cpp:240   INFO| Device stopped
21:07:24.794 [main thread     ]     SourceSelector.cpp:38    INFO| FdSource stopped
21:07:24.795 [main thread     ]           FdSource.cpp:53    WARN| Operation canceled
21:07:24.795 [main thread     ]           AlsaSink.cpp:240   INFO| Device stopped
21:07:24.795 [main thread     ]     SourceSelector.cpp:38    INFO| FdSource stopped
Transport state: BluezQt::MediaTransport::State::Pending
operator() fd:  13 mtu read: 672 mtu write: 1024
21:07:28.680 [main thread     ]        Persistence.cpp:63    INFO| filter count: 0
21:07:28.682 [main thread     ]       AudioManager.cpp:172   INFO| set transport> fd: 14, blocksize: 672, rate: 44100
Transport state: BluezQt::MediaTransport::State::Active
21:07:28.815 [main thread     ]     SourceSelector.cpp:52    INFO| Starting FdSource
21:07:28.815 [main thread     ]         RtpDecoder.cpp:67    INFO| sequence starts at: 466
21:07:28.816 [main thread     ]             Buffer.cpp:108   INFO| Buffer reallocated. 6204 -> 12408 bytes
21:07:28.822 [main thread     ]           AlsaSink.cpp:176   INFO| Device opened. delay: 100 ms, buffer: 500 ms
21:07:30.824 [main thread     ]           AlsaSink.cpp:281   WARN| Write failed: Broken pipe
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
21:07:32.138 [main thread     ]           AlsaSink.cpp:281   WARN| Write failed: Broken pipe
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
21:07:34.885 [main thread     ]           FdSource.cpp:53    WARN| End of file
21:07:34.891 [main thread     ]           AlsaSink.cpp:240   INFO| Device stopped
21:07:34.891 [main thread     ]     SourceSelector.cpp:38    INFO| FdSource stopped
21:07:34.894 [main thread     ]        Persistence.cpp:63    INFO| filter count: 0
21:07:34.895 [main thread     ]       AudioManager.cpp:172   INFO| set transport> fd: -1, blocksize: 0, rate: 0
21:07:34.896 [main thread     ]           AlsaSink.cpp:240   INFO| Device stopped
21:07:34.896 [main thread     ]     SourceSelector.cpp:38    INFO| FdSource stopped
```
Here it disconnected after a few seconds but I also had times where it took a bit longer to disconnect (but still with unusable audio quality). At that times it just spammed the whole time
```
AlsaSink.cpp:281   WARN| Write failed: Broken pipe
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
```

mincequi, 2024-04-20:
Maybe your phone uses an unsupported codec. Could you try to send SBC, Stereo, 16Bit and 44.1 or 48 kHz from your phone?

Blaubeeree, 2024-04-21:
The codec bitrate and frequency were already set to the right value, but I played around with the other bluetooth options I found and it seems to work when I deaktivate A2DP-Hardware-Offload in the Android developer options.

Thanks for your help!

closed 2024-04-21 by Blaubeeree