mincequi/cornrow #37: Support ffmpeg 7.0

KuabeM, 2024-12-09:
ffmpeg 7.0 was released in April 2024 and has made its way into Ubuntu 24.10, Fedora 41 etc. There are few API changes like "the removal of the old bitmask-based channel layout API, replaced by the AVChannelLayout" (see the news on the release on here https://www.ffmpeg.org/index.html#news, it's the only reference I really found). This API is used in the `AudioEncoderFfmpeg` and `AudioDecoderFfmpeg` classes.

In theory, these changes shouldn't be breaking (`AvChannelLayout` exists since 6.0) for Ubuntu 24.04 and Debian testing (trixie, sid, experimental) and should 'just work'. Previous Ubuntu versions up to 22.04 will probably break, though.

Question: are you open to supporting the new version and cutting off support for older distributions? Alternatively, one would support both versions and do some cmake magic to compile in the correct API depending on the version. I don't really have experience with this.

I'll open a PR with the changes in a minute, just for reference.

PS: I stumbled on this project a fews a day and already like it very much. Thanks you!

mincequi, 2024-12-10:
> Question: are you open to supporting the new version and cutting off support for older distributions? Alternatively, one would support both versions and do some cmake magic to compile in the correct API depending on the version. I don't really have experience with this.

Yes, i am open to remove support for older distributions. So, no cmake magic and ifdef hell is needed 🙂

> For referene, here's the PR with the required changes: [mincequi/libcornrow#1](https://github.com/mincequi/libcornrow/pull/1)
> 
> PS: I stumbled on this project a fews a day and already like it very much. Thanks you!

Nice, that you directly provide a PR to fix that. I will merge that.
Maybe, we should consider to make a release for libcornrow and add that via [CPM](https://github.com/cpm-cmake/CPM.cmake) to cornrow instead of a git submodule.

KuabeM, 2024-12-15:
Nice, thanks! Works like a charm :)

closed 2024-12-15 by KuabeM