mincequi/cornrow #41: cornrow will no longer pair with a device which has removed it

i-h8-github, 2024-12-19:
Hello,

When a device pairs with cornrow, then "forgets" cornrow, then tries to pair with it again, it becomes impossible due to "incorrect pairing code". This was tested on an Android phone.

Thank you for your software.

mincequi, 2024-12-22:
Hello,
not quite sure about your setup, but this might be an issue with the pairing policy of your system.

i-h8-github, 2024-12-23:
I guess it's whatever the pairing policy of an Android phone is. That phone is now permanently unable to pair with cornrow, regardless of any number of restarts. Only thing I haven't tried is factory resetting the phone.

I don't know what else there is to try with cornrow except restarting, but that also has no effect. I tried looking around the system for some kind of file related to cornrow to see if removing that could have an effect, but couldn't find any.

KuabeM, 2024-12-25:
Can you check if it's still remembered by the system running cornrow? I guess it's rather a problem that this system still knows about this device.

I usually manage bluetooth devices with `bluetoothctl`:

```bash
bluetoothctl devices
bluetoothctl remove <device id>
```

i-h8-github, 2025-01-05:
The phone still won't pair after having removed it via `bluetoothctl`.

KuabeM, 2025-01-05:
You could try connecting/paring your phone from the cornrow-system also via `bluetoothctl`. The comands should be along something like this:

```bash
# enter interactive menu
bluetoothctl
$ scan on
# scanning for discoverable/pairable devices, wait a bit until your phone shows up
$ devices
$ pair <mac-address>
$ trust <mac-address> # enable auto-connect
```

Make sure that your phone is discoverable.

You can also check the logs, maybe `journalctl -fu bluetooth.service` shows something relevant when you try to connect.

juhaj, 2025-03-12:
I am not the original reporter, but I can confirm the same symptom: after removing the cornrow device from Android, it is no longer possible to pair with it. Android (14, for what it's worth) keeps complaining the PIN or passkey is wrong.
The instructions given by @KuabeM resolve the issue:
```
bluetoothctl devices
bluetoothctl remove <device id>
```

i-h8-github, 2025-03-14:
Unfortunately, neither those instructions nor the later suggestion worked for me. I haven't looked at this in a while, but I will report back with any new findings soon.

mincequi, 2025-03-14:
Unfortunately, i cannot help here. Did anyone try if this can be reproduced with a different bluetooth audio implementation?

ea5522, 2025-03-27:
I was able to fix this by editing /etc/bluetooth/main.conf file and adding these parameters:

FastConnectable = true
JustWorksRepairing = always

If you remove/forget phone on android, you can connect back to it by just hitting "pair". You can also do the same from PC. If one device is connected, other devices will fail to connect. It fixed the pairing issue permanently. 

Please give it a try and let me know how it goes

i-h8-github, 2025-06-08:
> I was able to fix this by editing /etc/bluetooth/main.conf file and adding these parameters:
>
> FastConnectable = true JustWorksRepairing = always
>
> If you remove/forget phone on android, you can [...]

This seems to be doing the trick. Thank you very kindly for this tip.