mincequi/cornrow #43: Issues with sound in snapcast

ea5522, 2025-03-09:
I finally got it working on debian, but the audio quality is bad. What could be the issue?

mincequi, 2025-03-14:
Without further context or detailed description, it is hard to guess what is wrong. Is the audio choppy? Gaps? Noisy?

ea5522, 2025-03-14:
I will make a recording this weekend and post it

ea5522, 2025-03-26:
The audio is choppy. I tried to record audio clip using my phone, but the music stopped when I hit record, so I was unable to. Therefore, I decided to connect my laptop to Bluetooth and record using my phone, and the quality on the laptop is amazing! The Bluetooth quality is just as good as the "scream" network sound driver. It is definitely an android issue.  I'm using a Galaxy s24 

Here are the logs when using android phone:

Mar 25 23:46:39 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 3.97285 ms
Mar 25 23:46:39 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 3.83372 ms
Mar 25 23:46:39 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 3.89164 ms
Mar 25 23:46:39 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 4.42176 ms
Mar 25 23:46:39 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 2.86939 ms
Mar 25 23:46:39 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 1.87768 ms
Mar 25 23:46:39 snapcast snapserver[21992]: (StreamSessionTCP) Error reading message header of length 0: End of file
Mar 25 23:46:39 snapcast snapserver[21992]: (StreamServer) onDisconnect: cbcbbd03-399b-435a-b9e3-a79343ec28cd
Mar 25 23:46:39 snapcast snapserver[21992]: (StreamSession) StreamSession write error (msg length: 1856): Broken pipe
Mar 25 23:46:39 snapcast snapserver[21992]: (StreamSessionTCP) Error in socket shutdown: Transport endpoint is not connected
Mar 25 23:46:39 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 7.97276 ms
Mar 25 23:46:39 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 3.98201 ms
Mar 25 23:46:40 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 7.89103 ms
Mar 25 23:46:40 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 7.79222 ms
Mar 25 23:46:40 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 3.88192 ms
Mar 25 23:46:40 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 7.59925 ms
Mar 25 23:46:40 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 1.60733 ms
Mar 25 23:46:40 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 3.72871 ms
Mar 25 23:46:40 snapcast snapserver[21992]: (StreamServer) StreamServer::NewConnection: ::ffff:10.0.0.127
Mar 25 23:46:40 snapcast snapserver[21992]: (Server) Hello from cbcbbd03-399b-435a-b9e3-a79343ec28cd, host: SM-J737V, v0.29.0, ClientName: Snapclient, OS: Android 9, Arch: armeabi-v7a, Protocol version: 2
Mar 25 23:46:40 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 3.77409 ms
Mar 25 23:46:40 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 2.96053 ms
Mar 25 23:46:40 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 7.68846 ms
Mar 25 23:46:40 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 4.05797 ms
Mar 25 23:46:40 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 3.47127 ms
Mar 25 23:46:40 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 22.4954 ms
Mar 25 23:46:41 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 3.82982 ms
Mar 25 23:46:41 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 3.69355 ms
Mar 25 23:46:41 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 3.98289 ms
Mar 25 23:46:41 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 3.83451 ms
Mar 25 23:46:41 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 3.9397 ms
Mar 25 23:46:41 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 3.66831 ms
Mar 25 23:46:41 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 8.18101 ms
Mar 25 23:46:41 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 7.45795 ms
Mar 25 23:46:41 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 3.31532 ms
Mar 25 23:46:41 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 3.83082 ms
Mar 25 23:46:41 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 2.46077 ms
Mar 25 23:46:42 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 3.41143 ms
Mar 25 23:46:42 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 23.7203 ms
Mar 25 23:46:42 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 3.94791 ms
Mar 25 23:46:42 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 3.88117 ms
Mar 25 23:46:42 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 7.96651 ms
Mar 25 23:46:42 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 7.98373 ms
Mar 25 23:46:42 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 3.97171 ms
Mar 25 23:46:42 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 23.481 ms
Mar 25 23:46:42 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 3.60706 ms
Mar 25 23:46:43 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 3.77991 ms
Mar 25 23:46:43 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 3.6752 ms
Mar 25 23:46:43 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 3.64718 ms
Mar 25 23:46:43 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 7.8767 ms
Mar 25 23:46:43 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 0.821216 ms
Mar 25 23:46:43 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 8.01574 ms
Mar 25 23:46:43 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 3.64479 ms
Mar 25 23:46:43 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 22.6332 ms
Mar 25 23:46:43 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 3.90086 ms
[... cut ...]
Mar 25 23:46:45 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 4.31844 ms
Mar 25 23:46:45 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 3.51122 ms
Mar 25 23:46:45 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 3.80234 ms
Mar 25 23:46:45 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 7.66552 ms
Mar 25 23:46:45 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 3.86284 ms
Mar 25 23:46:45 snapcast snapserver[21992]: (Server) onResync (Bluetooth): 4.03619 ms

ea5522, 2025-03-26:
Thanks for that suggestion but I already tried playing with sample rates. 44.1 works best when I set it in snapcast. My Galaxy phone doesn't allow me to change it. I will try using using an iPhone or another android and will report back.

[image]